siliconcompiler/siliconcompiler #2062: Issue with Interpreting Outputs

cansudemirkiran, 2023-10-26:
Hi,

I have been testing the tool, and I have a few questions. When I feed different Verilog files into the tool or use different targets, even though I receive no errors, I sometimes can't collect all the output values, or they appear inconsistent. I use the '-remote' option for running jobs.

For example, when I run a Verilog file for a simple MAC unit with 'freepdk45' as the target, I can collect all the outputs from all the steps except 'export0'. When I run the same Verilog file with the 'asap7' target, I can complete the run without any errors, but I no longer get 'Fmax' and area information on the PNG file. The 'export 1' column also disappears.

It's also odd that the power numbers for the 7 nm library are more than ten times larger than the 45 nm numbers while running at a similar clock frequency. The 7 nm output shows a negative slack, I assume this means that the design does not meet the requirements. However, the clock period I defined in the constraint file was 10 ns, whereas the Fmax for this design appears to be over 1 GHz. Also, I expected the 7 nm design to be capable of running faster.

I've attached screenshots for these two cases, I'd greatly appreciate your feedback on how to interpret this. Thank you! 

[image]
[image]

gadfort, 2023-11-17:
@cansudemirkiran sorry for the delay. Both of these PDKs are virtual, so some of their behavior is going to be inconsistent. There are some setup differences between the two libraries (this is the only reason I can think of at the moment):
* freepdk45 only has a single timing and parasitic corner enabled (typical only), which will result in more ideal answers.
* asap7 has three corners enabled (slow, typical, and fast), so that would be one reason you end up seeing similar performance despite how much faster asap7 should have been.

gadfort, 2024-06-22:
Closing stale issue. Please let us know if you have any additional questions.

closed 2024-06-22 by gadfort