siliconcompiler/siliconcompiler #2080: Error about running the design flow of a buffer.

yyh-sjtu, 2023-11-08:
[image]
[image]
[image]
Excuse me, let me ask a question. As the pictures shown, this design realizes a buffer. Why the server reports error?

gadfort, 2023-11-17:
@yyh-sjtu sorry for the delay, a single buffer should not be a problem, but we seem to have some issues returning results reliably at the moment

gadfort, 2024-06-22:
I hope things became more stable and you were able to run your design. Please reopen or open a new issue if you are still encountering issues.

closed 2024-06-22 by gadfort