siliconcompiler/siliconcompiler #2108: Negligible Power Values

suhanimitra, 2023-11-26:
I have been trying to run a single-module Verilog file of mine but keep getting power values ~0 mw in the SiliconCompiler output. I've changed the period and tried numerous technology nodes (asap7, freepdk45, etc), but all are giving me the same issue. I get cellarea, memory, and other values but power doesn't have a value. I've also had this occur for other Verilog files of mine and am not sure how to address this. I have included the metrics table from this run below.

[image]

gadfort, 2023-11-29:
@suhanimitra I would check that you have a clock defined in your design. Without one, I think you would just get leakage power which is fairly small. The default units for the power is `mW`, which may be too large for a small design. Before the run begins you may want to switch the units to uW, by `chip.set('unit', 'power', 'uw')` (https://docs.siliconcompiler.com/en/latest/reference_manual/schema.html#param-unit-power-ref)

suhanimitra, 2023-12-01:
I had a clocked defined using `chip.clock('clk', period=1000)` in my python file. I also tried rerunning my file after changing the units of power using `chip.set('unit', 'power', 'uw')`, but am still getting the same power values as I initially did. I'm not sure if the unit change is working though, as the HTML file still shows the units as "mw". This is what my python file looks like:
[image]

gadfort, 2023-12-01:
I'm sorry, I was wrong, I meant to set the units for the metric itself.
`chip.set('metric', 'power', 'uw', field='unit')`

The units you are setting currently do not effect anything.
For asap7, a clock period of 1000ns is very large which will also contribute to the low power. You may want to consider lowering that some.

suhanimitra, 2023-12-01:
No worries!

I tried running it again, with `chip.set('metric', 'peakpower', 'uw', field='unit')` and a lower period of 100, but am only getting values of ~0.031 uw which is still quite small. Is there anything else that could be causing these power values to be so low?

gadfort, 2024-06-22:
Closing stale issue. If there is a specific improvement we can add you can open a separate issue for that.

closed 2024-06-22 by gadfort