siliconcompiler/siliconcompiler #2259: Mixed-Language Flow

sebinho, 2024-03-22:
Hi Guys,

I am new here and trying to get my hands on SiliconCompiler.
I am trying to run a simple design through SiliconCompiler. The particularity of my designs is that it contains mixed-language source files (VHDL, verilog, SV).

It is not very clear to me how to create a flow that can handle different languages for the same run.
Is this feasible and if so, is there anywhere I can find an example on how to achieve this?

Thanks for your help!

sebinho, 2024-03-22:
For the time being as a dirty hack, I call `ghdl`  within python with `subprocess`  to generate a verilog file from my VHDL sources.
I then add this file to the chip inputs together with the other verilog files I have in my design.

Not great but seems to work for now.

gadfort, 2024-03-22:
@sebinho it's not something that SC supports at the moment, but would be relatively "easy" to implement (I think). We just don't have a use case for it, if you are able to share yours or generate an example, I could take a look at implementing it as a first class feature.

At the moment, I would suggest creating a chip object for the VHDL code run it through convert and add the output to a second chip object that has the other code in it:

```
ghdl_chip = Chip('something')
ghdl_chip.input(...)
ghdl_chip.set('option', 'frontend', 'vhdl')
ghdl_chip.set('option', 'to', 'convert')
ghdl_chip.run()

verilog_chip = Chip('something')
verilog_chip .input(...)
verilog_chip .input(ghdl_chip.find_result('v', step='convert'))
...
```

sebinho, 2024-03-22:
@gadfort, many thanks for your proposal.
It is definitely much cleaner than what I have done so far using an external call.

I would guess that I will not be the only person interested in such a feature. Quite often you do get some legacy code that can be in a different language and hence mixed language support would be a huge added value to the flow of SiliconCompiler.

Much appreciated

sebinho, 2024-03-22:
@gadfort 

I had to slightly change your proposal to make it work. Here is the working version.
```python
ghdl_chip = Chip('something')
ghdl_chip.input(...)
flow = 'vlogconv'
ghdl_chip.node(flow, 'convert', convert)
ghdl_chip.set('option', 'flow', flow)
ghdl_chip.set('option', 'mode', 'asic')
ghdl_chip.run()

verilog_chip = Chip('something')
verilog_chip .input(...)
verilog_chip .input(ghdl_chip.find_result('v', step='convert'))
...
```
It seems mandatory to define a `flow` and a `mode`

closed 2024-04-25 by sebinho