siliconcompiler/siliconcompiler #2325: Error fetching remote results with freepdk45_demo, same design works in skywater130_demo

trdenton, 2024-04-07:
### Issue:
My example design fails with FileNotFoundError messages on **freepdk45_demo**.  However, it works without issue on **skywater130_demo.**, i.e. I do not see any error messages or stack traces.

My successful skywater130_demo run is invoked via `sc pe.sv pe.sdc -target "skywater130_demo" -remote`

### Environment:
I installed siliconcompiler via the quickstart guide/pip today (2024-04-07).  I'm on Ubuntu 22.04.1

my code is here: https://github.com/trdenton/ece1718_project_p2  

### Minimum steps to reproduce

 * Install siliconcompiler via pip
 * pull my git repo https://github.com/trdenton/ece1718_project_p2
 * run `sc pe.sv pe.sdc -target "freepdk45_demo" -remote`
 
### Troubleshooting attempted:

- with/without timescale directive in the .sv file
- have verified my .v file / testbench against iverilog -g1995 as well as -g2005 and -g2012
- have tested my .v file / testbench in modelsim (no issues or warnings)
- have tried renaming .sv file to .v
- tried with/without an sdc file
- have tried launching with a python script as well as with the `sc` command
- tried with/without the timescale directive
- tried with skywater130_demo (this works)

### Error output:
```
(venv) trdenton@lappy-t:~/src/ece1718_project_p2$ sc pe.sv pe.sdc -target "freepdk45_demo" -remote

 ____  _ _ _                    ____                      _ _
/ ___|(_) (_) ___ ___  _ __    / ___|___  _ __ ___  _ __ (_) | ___ _ __
\___ \| | | |/ __/ _ \| '_ \  | |   / _ \| '_ ` _ \| '_ \| | |/ _ \ '__|
 ___) | | | | (_| (_) | | | | | |__| (_) | | | | | | |_) | | |  __/ |
|____/|_|_|_|\___\___/|_| |_|  \____\___/|_| |_| |_| .__/|_|_|\___|_|
                                                   |_|

Authors: Andreas Olofsson, William Ransohoff, Noah Moroze, Zachary Yedidia, Massimiliano Giacometti, Kimia Talaei, Peter Gadfort, Aulihan Teng, Peter Grossmann, Gabriel Aguirre, Martin Troiber
Version: 0.21.7 

--------------------------------------------------------------------------------
| INFO    | pe.sv inferred as rtl/verilog
| INFO    | pe.sdc inferred as constraint/sdc
| INFO    | Command line argument entered: ['option', 'target'] Value: freepdk45_demo
| INFO    | Command line argument entered: ['option', 'remote'] Value: true
| WARNING | job0 | --        | - | Could not find remote server configuration: defaulting to https://server.siliconcompiler.com
| INFO    | job0 | --        | - | Setting up node import0 with surelog/parse
| INFO    | job0 | import | 0 | Tool 'surelog' found with version '1.82' in directory '/home/trdenton/src/ece1718_project_p2/sc/venv/lib/python3.10/site-packages/siliconcompiler/tools/surelog/bin'
| INFO    | job0 | import | 0 | Running in /home/trdenton/src/ece1718_project_p2/build/pe/job0/import/0
| INFO    | job0 | import | 0 | surelog -nocache -parse -nouhdm +libext+.sv+.v /home/trdenton/src/ece1718_project_p2/pe.sv -top pe
| INFO    | job0 | import | 0 | [INF:CM0023] Creating log file "/home/trdenton/src/ece1718_project_p2/build/pe/job0/import/0/slpp_all/surelog.log".
| INFO    | job0 | import | 0 | [WRN:PA0205] /home/trdenton/src/ece1718_project_p2/pe.sv:1:1: No timescale set for "pe".
| INFO    | job0 | import | 0 | [INF:CP0300] Compilation...
| INFO    | job0 | import | 0 | [INF:CP0303] /home/trdenton/src/ece1718_project_p2/pe.sv:1:1: Compile module "work@pe".
| INFO    | job0 | import | 0 | [INF:CP0302] Compile class "work@mailbox".
| INFO    | job0 | import | 0 | [INF:CP0302] Compile class "work@process".
| INFO    | job0 | import | 0 | [INF:CP0302] Compile class "work@semaphore".
| INFO    | job0 | import | 0 | [INF:EL0526] Design Elaboration...
| INFO    | job0 | import | 0 | [NTE:EL0503] /home/trdenton/src/ece1718_project_p2/pe.sv:1:1: Top level module "work@pe".
| INFO    | job0 | import | 0 | [NTE:EL0508] Nb Top level modules: 1.
| INFO    | job0 | import | 0 | [NTE:EL0509] Max instance depth: 1.
| INFO    | job0 | import | 0 | [NTE:EL0510] Nb instances: 1.
| INFO    | job0 | import | 0 | [NTE:EL0511] Nb leaf instances: 0.
| INFO    | job0 | import | 0 | [INF:UH0706] Creating UHDM Model...
| INFO    | job0 | import | 0 | [  FATAL] : 0
| INFO    | job0 | import | 0 | [ SYNTAX] : 0
| INFO    | job0 | import | 0 | [  ERROR] : 0
| INFO    | job0 | import | 0 | [WARNING] : 1
| INFO    | job0 | import | 0 | [   NOTE] : 5
| WARNING | job0 | import | 0 |  2: [WRN:PA0205] /home/trdenton/src/ece1718_project_p2/pe.sv:1:1: No timescale set for "pe".
| INFO    | job0 | import | 0 | Number of warnings: 1
| INFO    | job0 | import | 0 | Number of errors: 0
| INFO    | job0 | import | 0 | Finished task in 0.51s
| INFO    | job0 | --        | - | Collecting input sources
| INFO    | job0 | --        | - | Copying /home/trdenton/src/ece1718_project_p2/pe.sdc to '/home/trdenton/src/ece1718_project_p2/build/pe/job0/sc_collected_files' directory
| INFO    | job0 | --        | - | Copying /home/trdenton/src/ece1718_project_p2/pe.sv to '/home/trdenton/src/ece1718_project_p2/build/pe/job0/sc_collected_files' directory
| INFO    | job0 | remote    | 0 | Terms of Service info for this server:
| INFO    | job0 | remote    | 0 | -------------------------------------------------------------------------------------
| INFO    | job0 | remote    | 0 |   DISCLAIMER:
| INFO    | job0 | remote    | 0 |   - The open SiliconCompiler remote server is a free service. Please don't abuse it.
| INFO    | job0 | remote    | 0 |   - Submitted designs must be open source. SiliconCompiler is not responsible for any
| INFO    | job0 | remote    | 0 |     proprietary IP that may be uploaded.
| INFO    | job0 | remote    | 0 |   - Only send one run at a time (or you may be temporarily blocked).
| INFO    | job0 | remote    | 0 |   - Do not send large designs (machines have limited resources).
| INFO    | job0 | remote    | 0 |   - Nontrivial designs may take some time to run. (machines have limited resources).
| INFO    | job0 | remote    | 0 |   - For full TOS, see https://www.siliconcompiler.com/terms-of-service
| INFO    | job0 | remote    | 0 | -------------------------------------------------------------------------------------
| INFO    | job0 | remote    | 0 | Your job will be uploaded to a public server in 5 seconds.
| INFO    | job0 | remote    | 0 | Job started. Progress will be checked and reported every 30 seconds.
| INFO    | job0 | remote    | 0 | Your job's reference ID is: dba6395188ec47e181234218c4c7234d
| INFO    | job0 | remote    | 0 | Job is still running. Status:
| INFO    | job0 | remote    | 0 |   Running (1):
| INFO    | job0 | remote    | 0 |     syn0 (00:00:16)
| INFO    | job0 | remote    | 0 |   Pending (7): floorplan0, place0, cts0, ...
| INFO    | job0 | remote    | 0 | Job is still running. Status:
| INFO    | job0 | remote    | 0 |   Completed (1): syn0
| INFO    | job0 | remote    | 0 |   Running (1):
| INFO    | job0 | remote    | 0 |     floorplan0 (00:00:16)
| INFO    | job0 | remote    | 0 |   Pending (6): place0, cts0, route0, ...
| INFO    | job0 | remote    | 0 |   Fetching completed results:
| INFO    | job0 | remote    | 0 |     syn0
| INFO    | job0 | remote    | 0 | Job is still running. Status:
| INFO    | job0 | remote    | 0 |   Completed (3): syn0, floorplan0, place0
| INFO    | job0 | remote    | 0 |   Pending (5): cts0, route0, dfm0, ...
| INFO    | job0 | remote    | 0 |   Fetching completed results:
| INFO    | job0 | remote    | 0 |     floorplan0
| INFO    | job0 | remote    | 0 |     place0
| INFO    | job0 | remote    | 0 | Remote job completed! Retrieving final results...
Traceback (most recent call last):
  File "/home/trdenton/src/ece1718_project_p2/sc/venv/bin/sc", line 8, in <module>
    sys.exit(main())
  File "/home/trdenton/src/ece1718_project_p2/sc/venv/lib/python3.10/site-packages/siliconcompiler/apps/sc.py", line 75, in main
    chip.run()
  File "/home/trdenton/src/ece1718_project_p2/sc/venv/lib/python3.10/site-packages/siliconcompiler/core.py", line 3971, in run
    self._finalize_run(set(self._get_execution_exit_nodes(flow)), environment, status)
  File "/home/trdenton/src/ece1718_project_p2/sc/venv/lib/python3.10/site-packages/siliconcompiler/core.py", line 3647, in _finalize_run
    os.listdir(os.path.dirname(lastdir))
FileNotFoundError: [Errno 2] No such file or directory: '/home/trdenton/src/ece1718_project_p2/build/pe/job0/export'
```

trdenton, 2024-04-08:
note  - I have looked through the logs for error messages, this is what seems relevant:

```
(venv) trdenton@lappy-t:~/src/ece1718_project_p2/build/pe/job0$ find . -name "*log" -exec grep -Hin error '{}' \;
./place/0/place.log:101:[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
./place/0/place.log:102:Error: sc_place.tcl, 24 GPL-0307
```

the rest of the output seems unrelated: 
```
./import/0/import.log:17:[  ERROR] : 0
./import/0/slpp_all/surelog.log:29:[  ERROR] : 0
./floorplan/0/floorplan.log:92:    utl::error FLW 1 "Design contains unplaced IOs"
./floorplan/0/floorplan.log:110:        utl::error FLW 1 "Pin $name has placement specified in constraints, but also order."
./floorplan/0/floorplan.log:117:        utl::error FLW 1 "Pin $name doesn't have enough information to perform placement."
./floorplan/0/floorplan.log:141:        utl::error FLW 1 "Side number ($side) is not supported."
./floorplan/0/floorplan.log:145:    utl::error FLW 1 "$name needs to either specify side or layer parameter."
./floorplan/0/floorplan.log:175:        utl::error FLW 1 "Side number ($side) is not supported."
./floorplan/0/floorplan.log:179:    utl::error FLW 1 "$name needs to either specify side or layer parameter."
./floorplan/0/floorplan.log:190:      utl::error FLW 1 "Side number ($side) is not supported."
./floorplan/0/floorplan.log:214:        utl::error FLW 1 "Side number ($side) is not supported."
./floorplan/0/floorplan.log:298:        utl::error FLW 1 "Unable to create instance for $name as the cell has not been specified"
./floorplan/0/floorplan.log:302:          utl::error FLW 1 "Unable to create $name, $cell is not a valid type"
./floorplan/0/floorplan.log:329:# since we get an error otherwise.
./floorplan/0/floorplan.log:365:  utl::error FLW 1 "Design contains unplaced macros."
```

gadfort, 2024-04-08:
@trdenton The `FileNotFoundError: [Errno 2] No such file or directory` error should have been fixed in #2321 (I have not generated a new release to make that available yet).
The max_phi_cof relates to some numerical instability in openroad (I have not looked at the logs), but if the design is very small, OpenROAD sometimes cannot handle the placement. You can try adjusting the initial density: https://docs.siliconcompiler.com/en/latest/reference_manual/schema.html#param-constraint-density-ref
I'll take a closer look tomorrow.

gadfort, 2024-04-09:
@trdenton taking a look at the run, it looks like the design only has a few gates in it, which is probably why openroad failed.

trdenton, 2024-04-10:
@gadfort Thank you for looking into it

I've just tried running it from a MacOS machine instead and it magically worked?? job reference ID is b3ac086827274029a5f4908b202d2c86

(I followed the same process)

gadfort, 2024-04-15:
@trdenton are things back to functional for you?

trdenton, 2024-04-17:
@gadfort my use case is now working - part of the problem was a design issue that caused yosys to optimize away much of my design.  With this fixed, everything is working as expected on ubuntu 22.04 and macos sonoma

trdenton, 2024-04-17:
Marking as closed, python exception issue resolved for freepdk45_demo

closed 2024-04-17 by trdenton