siliconcompiler/siliconcompiler #2344: Add [source] tags to Docs linking to github source code

StephenMoreOSU, 2024-04-14:
I've been attempting to understand the SC source code and I find it difficult to find the source code related to API calls in the documentation. 

Additionally, because of the often dynamic nature of classes in SC python linters don't work which would let me track function calls back to their source code more easily.

For example if looking at the following function:
[SC Docs Chip.input](https://docs.siliconcompiler.com/en/stable/reference_manual/core_api.html#siliconcompiler.Chip.input)

It would be convenient to have a [source] tag following the function header which would link to the github source code and send me to the relevant lines.

Desired functionality can be seen here in the matplotlib docs:
[Matplotlib Axes Plot Docs](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.plot.html#matplotlib.axes.Axes.plot)

What I've been doing as an alternative is to just use a debugger and dive into functions at runtime but its not ideal.

gadfort, 2024-04-15:
@StephenMoreOSU thanks for raising this. I agree that should be present to make it easier to follow the code.

closed 2024-04-15 by gadfort

StephenMoreOSU, 2024-04-15:
Thanks so much for your quick response and PR!

PR #2345 (2024-04-15): add source link to docs
Closes #2344