siliconcompiler/siliconcompiler #2365: Default vias in OpenRoad

sebinho, 2024-04-22:
Hi,

I was wondering if there is any plan to support specifying default vias for the routing tool of OpenRoad?
I understand that there is currently not much support in OpenRoad to be able to use multi-cut vias and I actually posted a comment on this OpenRoad topic:
https://github.com/The-OpenROAD-Project/OpenROAD/issues/3071

Not sure how well TritonRouter will handle specifying different default vias compared to using default single cut vias.

I see that in SiliconCompiler, there is a placeholder for a setting regarding non default vias. Is this something that is working well? I will give it a shot but I suppose there might be some issues with more complex via rules such as the ones I have in my proprietary PDK. I also assume that the router will not be able to revert to another via (for example a single cut via) in case it cannot place the default double cut via.

@gadfort, did you guys every play with double cut vias and openroad?
Thanks a lot

gadfort, 2024-04-22:
You can with '[detailed_route_default_via']: https://docs.siliconcompiler.com/en/latest/reference_manual/schema.html#param-tool-task-var-ref): https://docs.siliconcompiler.com/en/latest/reference_manual/predef_modules/tools.html#tools-openroad-route-params-var-ref

You can specify the vias in the pdk setup under (https://docs.siliconcompiler.com/en/latest/reference_manual/schema.html#param-pdk-var-ref) with the tool set to `openroad` and the var is `detailed_route_default_via`

OpenROAD does not support redundant via insertion and will only use one via while routing.

sebinho, 2024-04-22:
@gadfort Thanks for the links.
I was able to get it running, but the existing code was not working. I opened a pull request here:
https://github.com/siliconcompiler/siliconcompiler/pull/2366

closed 2024-04-25 by sebinho