siliconcompiler/siliconcompiler #2382: STA after synthesis run

sebinho, 2024-04-28:
Hi,

From what I can gather, there isn't any timing report post-synthesis. The timing report seems to only come once we go through the floorplanning step.
This is often interesting to get some timing information early in the design phases. I see this is for example done in the OpenLane project where they do openSTA post synthesis and post routing.

I am open to contribute and write some scripts for this but I was wondering if there is any plan or progress on this matter @gadfort?

Many thanks for your time.

gadfort, 2024-04-29:
@sebinho it's a good idea, for the post route, the export task technically accomplishes this.  I don't have any plans to implement this, so if you have the time, I would be happy to review the PR.
I would suggest starting with just an openroad task, adding an opensta driver would be nice, but also a little bit more work.

sebinho, 2024-04-29:
@gadfort, I created a pull request with all my changes. It is working fine for me, maybe there are things that you would like to change so please take a look at it.
The only thing that is not perfect is that it reports a utilization of -100% due to the fact that at that stage there isn't any core/die area defined. If you you have a workaround, please let me know.
Here is the pull request: https://github.com/siliconcompiler/siliconcompiler/pull/2389

PR #2466 (2024-05-31): add opensta and staflow
Adds:
- `opensta` as a tool with the a single `timing` task.
- `staflow` a basic synthesis and static timing flow (frontend -> yosys -> opensta), I'm open to suggestions for the name of this flow.

Fixes:
- docs build to allow for tool drivers with `__init__.py`

Replaces: https://github.com/siliconcompiler/siliconcompiler/pull/2389
Closes: #2382, we can always revisit adding the timing step to the `asicflow`, but since the `floorplan` task already generating timing, it seemed like adding a separate flow would allow for better flexibility
closed 2024-05-31 by gadfort