siliconcompiler/siliconcompiler #2469: Issue with openroad version

athmanar, 2024-05-31:
### Version

0.22.0

### On which OS did this happen?

Linux

### On which Python version did this happen?

Python 3.8

### Reproduction Steps

```
bash siliconcompiler/setup/ubuntu22/install-openroad.sh
sc -target asic_demo # no remote option
```

```
| WARNING | job0 | floorplan | 0 | Version check on openroad failed with code 127
| INFO    | job0 | floorplan | 0 | Tool 'openroad' found with version 'directory' in directory '/usr/local/bin'
| ERROR   | job0 | floorplan | 0 | Version check failed for openroad. Check installation.
| ERROR   | job0 | floorplan | 0 | Found version directory, did not satisfy any version specifier set >=v2.0-13145.
| ERROR   | job0 | floorplan | 0 | Halting step 'floorplan' index '0' due to errors.
```

### Expected Behavior

show the summary

### Actual Behavior

error  as below:
```
| WARNING | job0 | floorplan | 0 | Version check on openroad failed with code 127
| INFO    | job0 | floorplan | 0 | Tool 'openroad' found with version 'directory' in directory '/usr/local/bin'
| ERROR   | job0 | floorplan | 0 | Version check failed for openroad. Check installation.
| ERROR   | job0 | floorplan | 0 | Found version directory, did not satisfy any version specifier set >=v2.0-13145.
| ERROR   | job0 | floorplan | 0 | Halting step 'floorplan' index '0' due to errors.
```

gadfort, 2024-05-31:
@athmanar can you try running: `openroad -version` to verify the installation and report back the output of the command?

athmanar, 2024-05-31:
> p

actually when i do this:
```
/usr/local/bin/openroad -version
or 
openroad 
```

I get the following error:
`/usr/local/bin/openroad: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory`

gadfort, 2024-05-31:
It seems that the OpenROAD install is somehow incomplete. I would suggest filing an issue with the OpenROAD team.
In the `deps/openroad` folder you should be able to run `./etc/Env.sh` and generate the information the would need to be able to to help you. File the ticket at: https://github.com/The-OpenROAD-Project/OpenROAD/issues/new?assignees=&labels=&projects=&template=bug_report.yml

closed 2024-05-31 by athmanar