siliconcompiler/siliconcompiler #2582: Cyclic PIP dependencies

hpretl, 2024-06-28:
### Version

0.23.3

### On which OS did this happen?

Linux

### On which Python version did this happen?

Python 3.10

### Reproduction Steps

I am using `pipdeptree`, it throws the following warning:

```
Warning!!! Cyclic dependencies found:
* siliconcompiler => lambdapdk => siliconcompiler
* lambdapdk => siliconcompiler => lambdapdk
```

### Expected Behavior

Not sure if really problemtic, but proably not nice either.

### Actual Behavior

No cyclic redundancy, I guess?

gadfort, 2024-06-28:
Circular dependencies are permitted: https://discuss.python.org/t/handling-installation-of-circular-dependencies/25531/6
While not great, as long as the version restrictions do not cause a failure (in both projects we have them set to `>=` to avoid a failure in installation).
If you are having an actual issue with an install we can look at fixing that.

hpretl, 2024-06-28:
No issue, just wanted to let you know.

closed 2024-06-28 by hpretl