siliconcompiler/siliconcompiler #2642: Tutorial / Advice for using commercial CAD tools (e.g. Synopsys Design Compiler, Cadence Innovus, Synopsys Primetime)

StephenMoreOSU, 2024-07-11:
### Feature Description

I'm interested in running designs through sc using standard commercial ASIC tools (in title). 

##Steps Taken

* I did a few `grep -R` text searches on the repo looking for "synopsys", "cadence", "design compiler" and the only place I could find them were in the docs under [supported technologies](https://docs.siliconcompiler.com/en/stable/user_guide/what_is_sc.html#supported-technologies). 
* Looked through the FAQ and tools documentation
* Looked through examples

If there is any information on how to connect commercial CAD tools to sc it would be interesting. 

Thanks

gadfort, 2024-07-19:
@StephenMoreOSU due to license limitations, we do not have the synopsys and cadence enablement available publicly.
To add a tool you would need to add a tool and task driver (you can see how that is done with yosys or openroad), and then you would need to add a flow that makes use of these tasks. We probably need a proper tutorial for adding a tool.

infinitymdm, 2025-04-28:
Would contributions of tool/task drivers for proprietary tools be accepted? My team is interested in using siliconcompiler with Synopsys tools. As far as I'm aware, we're under no such license restrictions, but I don't want to make things complicated for your group.

Blebowski, 2025-07-15:
Hello @gadfort, is this something that is true for all SNPS / CDNS tools ? 

E.g. the OSVVM (VHDL simulation framework) supports SNPS VCS and Cadence Xcelium.
Sure, there it only invokes the simulator, but part of the command set for controlling the simulator
is exposed publicly...

Sure one can't expect that a whole TCL command set of a tool like SNPS DC can be exposed,
but maybe the basics such as `read_vhdl` , `read_verilog`, `elaborate`, `link`, `target_libraries` etc... 
are OK to be exposed, and mapped to the Silicon compiler interfaces (e.g. by pre-generating some tool specific TCL).

Did you happen to have this discussion with SNPS or CDNS ? I would be curious about it.

I would like to try to use the Silicon Compiler at the company where I work, but unless there 
is a good support of at least VCS and DC, I would need to write a lot to get it running.

For the same purpose as SC exists, we developed our own internal python flow for front-end simulation + synthesis,
having own "source list files", "test list files", "PDK view structures", and ways to export paths to these
files in various forms to be digested by the EDA tools. The amount of work we invested into it is significant,
and the project has lot of tech-debt. Definitely SC has better architecture, and it would make sense for
us to switch to it in the long run. However, if we need to maintain own "internal plugins" just to plug in the
commercial EDA tools we use, IMHO the effort is the same as maintaining own internal flow.

gadfort, 2025-07-21:
@Blebowski The support we have for synopsys cannot be shared unfortunately. You can talk to your representatives at the companies to see what they will allow.
I agree that it would be best if the community could share a common set of enablements for the tools, but the current licenses do not permit that.