siliconcompiler/siliconcompiler #2881: sc-install local installation KeyError

StephenMoreOSU, 2024-10-09:
### Version

0.28.2

### On which OS did this happen?

Linux

### On which Python version did this happen?

Python 3.12

### Reproduction Steps

All commands run:

```bash
git -b v0.28.2 git@github.com:siliconcompiler/siliconcompiler.git
cd siliconcompiler
python3 -m venv ./sc-venv
source ./sc-venv/bin/activate
python3 -m pip install -e .
sc-install -group asic
```

### Expected Behavior

Open source ASIC tool dependencies should be installed locally on my system.

### Actual Behavior

## Receiving the following error

Traceback (most recent call last):
  File "~/open-source-cad/siliconcompiler/sc-venv/bin/sc-install", line 8, in <module>                                                   
    sys.exit(main())
             ^^^^^^
  File "~/open-source-cad/siliconcompiler/siliconcompiler/apps/sc_install.py", line 185, in main                                         
    if not install_tool(tool, tools[tool], args.build_dir, args.prefix):
                              ~~~~~^^^^^^
KeyError: 'yosys'

- Note: In the original error the file paths were returned as absolute; however, I replaced them with `~` to obfuscate my system paths.

gadfort, 2024-10-09:
@StephenMoreOSU what OS are you using?
I see the bug and I can fix it, but it's possible we dont have build scripts for your specific OS.

StephenMoreOSU, 2024-10-09:
Ah I see, my OS is Rocky Linux 8.1

gadfort, 2024-10-09:
That should be supported.
Could you run: 
```
from siliconcompiler.scheduler import _get_machine_info
machine_info = _get_machine_info()

print("System:   ", machine_info.get('system', None))
print("Distro:   ", machine_info.get('distro', None))
print("Version:  ", machine_info.get('osversion', None))
```

And let me know what is says?

StephenMoreOSU, 2024-10-09:
Output I'm getting is:
```plaintext
System:    linux
Distro:    rocky
Version:   8.10
```

gadfort, 2024-10-09:
Ahhh, I see I got myself confused.
The rhel8/rocky8 scripts were added in https://github.com/siliconcompiler/siliconcompiler/pull/2872 which was after the last release.
So on the next release of 0.28.3 they should be included.

I'll fix the buggy behavior in `sc-install` and try to generate a release this week, but for now you can just move to the head of main to install.

gadfort, 2024-10-09:
I also noticed a slight bug in the script for yosys: https://github.com/siliconcompiler/siliconcompiler/pull/2883 so that should be fix now as well.

StephenMoreOSU, 2024-10-09:
I see, thanks for the fast responses!
I was actually able to install the tools in a docker container using files in [setup/docker](https://github.com/siliconcompiler/siliconcompiler/tree/main/setup/docker) and some minor hacking, not trying to add unnecessary work but the local docker method would be ideal for my use case in the future.

gadfort, 2024-10-10:
If all you need is the docker enablement, we do publish: https://github.com/siliconcompiler/siliconcompiler/pkgs/container/sc_tools/282607425?tag=latest
Which should contain all the tools.
Additionally, you can set the scheduler to `docker` and it will pull this image automatically and run the tools in docker.
https://docs.siliconcompiler.com/en/latest/user_guide/docker.html

g3grau, 2024-10-14:
I just came across the same on openSuse Tumbleweed. python3 -c "import
 ```
siliconcompiler;print(siliconcompiler.__version__)"
0.28.2
```
```
sc-install -group asic
Traceback (most recent call last):
  File "/usr/cadtools/miniconda3/envs/siliconcompiler/bin/sc-install", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/cadtools/miniconda3/envs/siliconcompiler/lib/python3.11/site-packages/siliconcompiler/apps/sc_install.py", line 185, in main
    if not install_tool(tool, tools[tool], args.build_dir, args.prefix):
                              ~~~~~^^^^^^
KeyError: 'sv2v'
```
```
sc-install -group fpga
Traceback (most recent call last):
  File "/usr/cadtools/miniconda3/envs/siliconcompiler/bin/sc-install", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/cadtools/miniconda3/envs/siliconcompiler/lib/python3.11/site-packages/siliconcompiler/apps/sc_install.py", line 185, in main
    if not install_tool(tool, tools[tool], args.build_dir, args.prefix):
                              ~~~~~^^^^^^
KeyError: 'surelog'
```
The OS test returns this:
```
Python 3.11.10 | packaged by conda-forge | (main, Sep 30 2024, 18:08:57) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from siliconcompiler.scheduler import _get_machine_info
>>> machine_info = _get_machine_info()
>>> 
>>> print("System:   ", machine_info.get('system', None))
System:    linux
>>> print("Distro:   ", machine_info.get('distro', None))
Distro:    opensuse-tumbleweed
>>> print("Version:  ", machine_info.get('osversion', None))
Version:   20241007
>>> 
```

gadfort, 2024-10-14:
https://github.com/siliconcompiler/siliconcompiler/releases/tag/v0.28.3

StephenMoreOSU, 2024-10-15:
I tried doing the local install on v0.28.3 and when running the following command
```bash
sc-install -group asic
```
I now get the following output:
```
unsupported system: Rocky Linux

Usage: ./etc/DependencyInstaller.sh
                                # Installs all of OpenROAD's dependencies no
                                #     need to run -base or -common. Requires
                                #     privileged access.
                                # ...
```

gadfort, 2024-10-15:
@StephenMoreOSU OpenROAD does not support rocky linux, I suggest you submit a ticket at https://github.com/The-OpenROAD-Project/OpenROAD/issues