siliconcompiler/siliconcompiler #2965: Add feature to select Yosys version

dkongg, 2024-11-05:
### Feature Description

Hello,
I have some RTL code that passes through the flow in Yosys 0.44. It seems like the Yosys version has been updated to 0.46. 
Now floorplanning fails with the following error: 12: [ERROR STA-0164] inputs/design.vg line 6, syntax error, unexpected REG.
However, floorplanning didn't fail when Yosys 0.44 was used for synthesis.

gadfort, 2024-11-05:
@dkongg you can always just install the needed version of yosys, we only compile and test against a single version.
It sounds like what you are really encountering is an issue with openroad/opensta. It might be helpful to have the testcase (if it can be shared) or just the anonymised lines from `inputs/design.vg` sounds like the first 7 lines would be sufficient.

dkongg, 2024-11-05:
@gadfort here are the first few lines of the .vg file. I think what's happening is that the tool is unable to implement when the first row the module is a reg instead of a wire. 

`/* Generated by Yosys 0.46 (git sha1 e97731b9d, g++ 9.4.0-1ubuntu1~20.04.2 -fPIC -O3) */
(* top =  1  *)
(* src = "inputs/design.v:82.1-172.10" *)
module design(clk, rst, en, plaintext, key, ciphertext, done);
  reg \$auto$verilog_backend.cc:2352:dump_module$58570  = 0;
  (* src = "inputs/design.v:0.0-0.0" *)
  wire [31:0] _00000_;
  (* src = "inputs/design.v:0.0-0.0" *)
  wire [31:0] _00001_;
  (* src = "inputs/design.v:0.0-0.0" *)

gadfort, 2024-11-05:
how was this generated? the `reg $auto$verilog_backend.cc:2352:dump_module$58570 = 0;` is highly unusual.
This seems like yosys failed to do something, since the output of yosys should only contain gates and wires, it cannot have registers since that should have been converted to a FF.
I assume this is not driving anything in the design?

dkongg, 2024-11-06:
I ran this remotely on the cloud server and the job ID is 329f111e73f348f39e86c1b5d1f9ea55. 
There are no synthesis errors but there is a synthesis warning saying:
"Use "proc" to convert processes to logic networks and registers."
Interestingly, the full flow was working when yosys 0.44 was used. Perhaps the "proc" command could be added in the synthesis script?

gadfort, 2024-11-06:
@dkongg it looks like yosys broke something, proc is already included, but looking at the logs it seems to fail on something.
Is the code sharable? If yes, I would like to generate a testcase for the yosys team so they can fix what broke.

dkongg, 2024-11-06:
The code is shareable, thank you.

gadfort, 2024-11-06:
I've opened an issue with yosys.
Looking at the log you also have this:
```
Warning: Async reset value `\key [31:0]' is not constant!
  created $aldff cell `$procdff$1503' with positive edge clock and positive level non-const reset.
```

which appears to unmapped also in the 0.44 version.

gadfort, 2024-11-07:
This fixes the issue: https://github.com/YosysHQ/yosys/pull/4714 you still have the `$_ALDFFE_PPP_` in the design which are not mapped correctly, but that issue appears to have been present since 0.44

dkongg, 2024-12-11:
Hi @gadfort, would it be possible to update the Yosys version to one that been fixed? (seems like the main branch on Yosys doesn't have the fix... https://github.com/YosysHQ/yosys/issues/4712#issuecomment-2493065287)

gadfort, 2024-12-11:
@dkongg the remote will update once I make a new release of SC. My hope is that will be today at some point. The version of yosys (yosys 0.48) they released today contains the fix (https://github.com/YosysHQ/yosys/pull/4714).

closed 2024-12-11 by gadfort