siliconcompiler/siliconcompiler #3062: Silicon Compiler Server not working

Subashkatel, 2024-12-12:
### Version

0.28.9

### On which OS did this happen?

macOS

### On which Python version did this happen?

Python 3.10

### Reproduction Steps

Add ```chip.set('option', 'remote', True)``` on heartbeat.py before line 13
Then run  python heartbeat.py example 

### Expected Behavior

[image]

### Actual Behavior

Its suppose to run the example and provide the output but its stuck on a loop

gadfort, 2024-12-12:
@Subashkatel I'll restart the runners and that should restart the queue, but you may need to resubmit your job since it looks like the jobqueue needs to be reset as well.

Subashkatel, 2024-12-12:
Okay thank you

gadfort, 2024-12-12:
I'll let you know once the runners are back up.

gadfort, 2024-12-12:
@Subashkatel runners should be back up and accepting jobs

closed 2024-12-13 by gadfort

dkongg, 2024-12-20:
Hello @gadfort, I am also unable to run the heartbeat example.
My job ID is f20844201a394953ade09403ff30d1a6 and python version is 3.11.11.
Could you look into the issue? Thanks

gadfort, 2024-12-20:
You will need to update to SC 0.29.0 for the server to accept the job. You can see the log (I think) in:
heartbeat/job0/configs/siliconcompiler.log

gadfort, 2024-12-20:
I've noticed some issues with power grid generation, which I'm taking a look at as well.