siliconcompiler/siliconcompiler #3088: Analog Design Flows

curtisma, 2025-01-01:
### Feature Description

Would there be interest in supporting analog design flows in SiliconCompiler?
Here are a few ideas to consider.  I was considering trying to use it on a simple open-source band-gap voltage reference design project to see if it would work well.

The most basic flow would be something like this:

Netlist -> Simulate ->  Post-process and/or Document results

The input file would point to a schematic.  We could start by supporting open-source [Xschem](https://xschem.sourceforge.io/stefan/xschem_man/xschem_man.html) schematics.

It would be great if it could take a CSV PVT "corners" conditions file and then run a set of simulations in parallel.

**Tasks and associated Open-source Tools that could support them:**
- Netlist using [Xschem](https://xschem.sourceforge.io/stefan/xschem_man/xschem_man.html)
- Save a schematic picture using Xschem
- Simulate using NgSpice and Xyce
- Report the results to a CSV file, Python post-processing script, or other method of documentation
- Support the [HDL21 Python "harware description library" ](https://github.com/dan-fritchman/Hdl21)for writing and netlisting analog test benches.

Hopefully a lot of the DRC and LVS tools and tasks would still apply, so verification of the layout would be similar, just with Xschem generating the Netlist for LVS.

aolofsson, 2025-01-03:
Yes!! It has just been a question of time and need (we do digital design almost exclusively).

- We have initial support of xyce, but the simulations so far have been from netlist. If the schematic is the source, then the netlisting should be automated along with the simulation.
- We do support PVT scenarios already for digital we should be able to use it for analog as well.
- We would definitely welcome an xschem plug in contribution! Here is the list of the current ports. The Verilator plug in is a good starting point. https://github.com/siliconcompiler/siliconcompiler/tree/main/siliconcompiler/tools
- I am not a big fan of CSV files in general, we have an extensive JSON schema entered via a python API.
- Do you have an official output format for the results (ie vcd for analog)?

curtisma, 2025-01-05:
Great, good to hear!

> * We would definitely welcome an xschem plug in contribution! Here is the list of the current ports. The Verilator plug in is a good starting point. https://github.com/siliconcompiler/siliconcompiler/tree/main/siliconcompiler/tools

I have a working driver for Xschem already working at:
https://github.com/cascode-labs/viper/blob/main/viper/schematics/XschemSchematic.py

It will just need to be converted to your tool and task format.

gadfort, 2025-01-06:
@curtisma let me know if you need any assistance, the driver you have looks good, so hopefully should not require much additional effort to convert it.