siliconcompiler/siliconcompiler #3140: Flowgraph race condition when setting up nodes

Ciprian167, 2025-01-20:
### Version

0.29.0

### On which OS did this happen?

Linux

### On which Python version did this happen?

Python 3.11

### Reproduction Steps

I'm using proprietary code so I can't share the manifest. I have a graph that looks like below. The behavior happens on `clean` runs, with `/build` directory removed for extra insurance. 

[image]

`xrun/compile` is a task that receives files from other nodes as `input` and generates a single file as `output`.  Its setup function looks something like this:

```python

def setup(chip):
    setup_tools(chip)

    step = chip.get("arg", "step")
    index = chip.get("arg", "index")
    tool, task = get_tool_task(chip, step, index)
    # ...
    chip.add("tool", tool, "task", task, "output", os.path.join(chip.getworkdir(step=step, index=index), "compiled.lib"), step=step, index=index)
    chip.set( "tool", tool, "task", task, "input", list(input_provides(chip, step, index).keys()), step=step, index=index)
    # ...
```

### Expected Behavior

I'm expecting node `compile_top_tb` to receive two files, one from `compile_ip2` and another one from `compile_uvm`.

### Actual Behavior

Sometimes, `compile_top_tb` receives the file only from `compile_uvm`, even though the flow graph shows the dependency between them.  

I added some prints in the `setup` function and displayed the input nodes and the files for each node. 
```text
| INFO    | custom_job | -----                | - | Setting up node compile_uvm0 with xrun/compile_simple
| WARNING | custom_job | -----                | - | Nodes: []
| INFO    | custom_job | -----                | - | Setting up node compile_common_ip with xrun/compile_simple
| WARNING | custom_job | -----                | - | Nodes: []
| INFO    | custom_job | -----                | - | Setting up node compile_ip10 with xrun/compile_simple
| WARNING | custom_job | -----                | - | Nodes: [('compile_common_ip', '0')]
| WARNING | custom_job | -----                | - | Input: ~/sc/build/mypkg/custom_job/compile_common_ip/0/compiled.lib
| INFO    | custom_job | -----                | - | Setting up node compile_top_tb0 with xrun/compile_simple
| WARNING | custom_job | -----                | - | Nodes: [('compile_ip2', '0'), ('compile_uvm', '0')]
| WARNING | custom_job | -----                | - | Input: ~/sc/build/mypkg/custom_job/compile_uvm/0/compiled.lib
| INFO    | custom_job | -----                | - | Setting up node compile_ip20 with xrun/compile_simple
| WARNING | custom_job | -----                | - | Nodes: [('compile_common_ip', '0'), ('compile_ip1', '0')]
| WARNING | custom_job | -----                | - | Input: ~/sc/build/mypkg/custom_job/compile_common_ip/0/compiled.lib
| WARNING | custom_job | -----                | - | Input: ~/sc/build/mypkg/custom_job/compile_ip1/0/compiled.lib
| INFO    | custom_job | -----                | - | Checking manifest before running.
```

Based on the log, you can see that node `compile_top_tb` has two input nodes, but receives only one file form `compile_uvm`. 

I suspect the problem is related to the order the nodes are being `setup`. Based on the log, we call `setup` for `compile_top_tb` before `compile_ip2`, this means that `compile_ip2` have not yet registered the output files to the "schema".

gadfort, 2025-01-20:
@Ciprian167 thanks, took a bit to reproduce, but I think I have it fixed.

As an aside, ideally you would not use abspaths in the input/output keys. A post processing step would be able to ensure the resultant files are in the `outputs/` directory, allowing the scheduler to handle the IOs.

closed 2025-01-20 by gadfort

Ciprian167, 2025-01-21:
> [@Ciprian167](https://github.com/Ciprian167) thanks, took a bit to reproduce, but I think I have it fixed.
>
> As an aside, ideally you would not use abspaths in the input/output keys. A post [...]

Maybe this is not the place but, a follow up on this comment. I'm not providing abspaths in the input and output keys directly. The tasks are working with/on libraries, and when I'm registering a file to a library I'm using relative paths. The abspath is begin used when I call `register_source`. Here's an example:

```
import os
import siliconcompiler

import src.lib1.lib1 as lib1
import src.lib2.lib2 as lib2

def setup():
    lib = siliconcompiler.Library("prim", package="prim", auto_enable=True)
    lib.register_source(name="prim", path=os.path.abspath(os.path.dirname(__file__)))

    lib.add("input", "rtl", "verilog", "prim.v")
    lib.add("input", "rtl", "verilog", "mod1.v")
    lib.add("input", "rtl", "verilog", "mod2.v")
    lib.add("input", "elab", "href", "config/href.txt")

    lib.use(lib1)
    lib.use(lib2)

    return lib
```

Is this the intended usage? Also, I don't see anything in the `inputs/ouputs/ dirs. Would they get populated when I use libraries as well?

Ciprian167, 2025-01-21:
FIY, I tried setting the `copy` parameter for the library the following way:

Right in the library's setup function:
```python
lib.set("input", "rtl", "systemverilog", True, field='copy')
lib.set("input", "rtl", "verilog", True, field='copy')
```
In the task itself like:
```python
chip.set("library", design_lib, "input", "rtl", "systemverilog", True, field='copy')
chip.set("library", design_lib, "input", "rtl", "verilog", True, field='copy')
```
where `design_lib` is the library being processed. 

None of these seem to generate the output you describe.

gadfort, 2025-01-27:
@Ciprian167 
I meant in the task driver:
```
chip.add("tool", tool, "task", task, "output", "compiled.lib", step=step, index=index)
chip.set( "tool", tool, "task", task, "input", <name>, step=step, index=index)
```

If you need to receive the same file from different nodes, they can be enumerated as:
```
chip.set( "tool", tool, "task", task, "input", "compiled.<step0><index>.lib", step=step, index=index)
chip.set( "tool", tool, "task", task, "input", "compiled.<step1><index>.lib", step=step, index=index)
chip.set( "tool", tool, "task", task, "input", "compiled.<step2><index>.lib", step=step, index=index)
```

PR #3141 (2025-01-20): fix execution order calculation to account for multiple input nodes f…
…rom different levels

Closes #3140