siliconcompiler/siliconcompiler #3209: Failed to compile Bluespec SystemVerilog design

chenshih1, 2025-02-13:
I attempted to compile Bluespec SystemVerilog using SiliconCompiler, but it failed. The built-in modules in bsc were not correctly imported, and the compilation log is as follows:
```
(venv) chens@chens-ThinkPad-E470c:~/workspace/Bluespec_BSV_Practice/DotProduct$ python dotproduct.py 
| INFO    | /home/chens/workspace/Bluespec_BSV_Practice/DotProduct/DotProduct.bsv inferred as hll/bsv
| INFO    | job0 | -----                | - | Setting up node import.bluespec0 with bluespec/convert
| INFO    | job0 | -----                | - | Setting up node import.c0 with bambu/convert
| WARNING | job0 | -----                | - | Removing import.c0 due to no files in [input,hll,c] or [input,hll,llvm]
| INFO    | job0 | -----                | - | Setting up node import.chisel0 with chisel/convert
| WARNING | job0 | -----                | - | Removing import.chisel0 due to no files in [input,hll,scala] or [input,config,chisel]
| INFO    | job0 | -----                | - | Setting up node import.verilog0 with surelog/parse
| WARNING | job0 | -----                | - | Removing import.verilog0 due to no files in [input,rtl,systemverilog] or [input,rtl,verilog]
| INFO    | job0 | -----                | - | Setting up node import.vhdl0 with ghdl/convert
| WARNING | job0 | -----                | - | Removing import.vhdl0 due to no files in [input,rtl,vhdl]
| INFO    | job0 | -----                | - | Setting up node import.convert0 with sv2v/convert
| WARNING | job0 | -----                | - | Removing import.convert0 due to will not receive systemverilog to convert
| INFO    | job0 | -----                | - | Setting up node import.combine0 with builtin/concatenate
| WARNING | job0 | -----                | - | Removing import.combine0 due to no need to concatenate file
| INFO    | job0 | -----                | - | Setting up node syn0 with yosys/syn_asic
| INFO    | job0 | -----                | - | Setting up node floorplan.init0 with openroad/init_floorplan
| INFO    | job0 | -----                | - | Setting up node floorplan.macro_placement0 with openroad/macro_placement
| INFO    | job0 | -----                | - | Setting up node floorplan.tapcell0 with openroad/endcap_tapcell_insertion
| INFO    | job0 | -----                | - | Setting up node floorplan.power_grid0 with openroad/power_grid
| INFO    | job0 | -----                | - | Setting up node floorplan.pin_placement0 with openroad/pin_placement
| INFO    | job0 | -----                | - | Setting up node place.global0 with openroad/global_placement
| INFO    | job0 | -----                | - | Setting up node place.repair_design0 with openroad/repair_design
| INFO    | job0 | -----                | - | Setting up node place.detailed0 with openroad/detailed_placement
| INFO    | job0 | -----                | - | Setting up node cts.clock_tree_synthesis0 with openroad/clock_tree_synthesis
| INFO    | job0 | -----                | - | Setting up node cts.repair_timing0 with openroad/repair_timing
| INFO    | job0 | -----                | - | Setting up node cts.fillcell0 with openroad/fillercell_insertion
| INFO    | job0 | -----                | - | Setting up node route.global0 with openroad/global_route
| INFO    | job0 | -----                | - | Setting up node route.antenna_repair0 with openroad/antenna_repair
| INFO    | job0 | -----                | - | Setting up node route.detailed0 with openroad/detailed_route
| INFO    | job0 | -----                | - | Setting up node dfm.metal_fill0 with openroad/fillmetal_insertion
| INFO    | job0 | -----                | - | Setting up node write.gds0 with klayout/export
| INFO    | job0 | -----                | - | Setting up node write.views0 with openroad/write_data
| INFO    | job0 | -----                | - | Checking manifest before running.
| INFO    | job0 | import.bluespec      | 0 | Tool 'bsc' found with version '2024.07' in directory '/home/chens/local/bsc/bin'
| INFO    | job0 | import.bluespec      | 0 | Running in /home/chens/workspace/Bluespec_BSV_Practice/DotProduct/build/mkDotProduct_nt_Int32/job0/import.bluespec/0
| INFO    | job0 | import.bluespec      | 0 | Found siliconcompiler data at /home/chens/venv/lib/python3.10/site-packages/siliconcompiler
| INFO    | job0 | import.bluespec      | 0 | bsc -verilog -vdir verilog -u -g mkDotProduct_nt_Int32 -p /home/chens/workspace/Bluespec_BSV_Practice/DotProduct:%/Libraries /home/chens/workspace/Bluespec_BSV_Practice/DotProduct/DotProduct.bsv
| INFO    | job0 | import.bluespec      | 0 | Finished task in 1.1s
| INFO    | job0 | syn                  | 0 | Found cached lambdapdk data at /home/chens/.sc/cache/lambdapdk-v0.1.48
| INFO    | job0 | syn                  | 0 | Tool 'yosys' found with version '0.50' in directory '/home/chens/local/bin'
| INFO    | job0 | syn                  | 0 | Running in /home/chens/workspace/Bluespec_BSV_Practice/DotProduct/build/mkDotProduct_nt_Int32/job0/syn/0
| INFO    | job0 | syn                  | 0 | yosys -c /home/chens/venv/lib/python3.10/site-packages/siliconcompiler/tools/yosys/sc_syn.tcl
| ERROR   | job0 | syn                  | 0 | 3.2. Analyzing design hierarchy..
| ERROR   | job0 | syn                  | 0 | Top module:  \mkDotProduct_nt_Int32
| ERROR   | job0 | syn                  | 0 | Removed 0 unused modules.
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | yosys> tribuf
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | 4. Executing TRIBUF pass.
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | yosys> scratchpad -set flatten.separator /
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | yosys> synth -flatten -extra-map /home/chens/venv/lib/python3.10/site-packages/siliconcompiler/tools/yosys/techmaps/lcu_kogge_stone.v -top mkDotProduct_nt_Int32 -run begin:fine
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | 5. Executing SYNTH pass.
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | yosys> hierarchy -check -top mkDotProduct_nt_Int32
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | 5.1. Executing HIERARCHY pass (managing design hierarchy).
| ERROR   | job0 | syn                  | 0 | 
| ERROR   | job0 | syn                  | 0 | 5.1.1. Analyzing design hierarchy..
| ERROR   | job0 | syn                  | 0 | ERROR: Module `\FIFO2' referenced in module `\mkDotProduct_nt_Int32' in cell `\dut_fifo_out' is not part of the design.
| WARNING | job0 | syn                  | 0 | Command failed with code 1. See log file /home/chens/workspace/Bluespec_BSV_Practice/DotProduct/build/mkDotProduct_nt_Int32/job0/syn/0/syn.log
| WARNING | job0 | syn                  | 0 | Yosys cell statistics are missing
| INFO    | job0 | syn                  | 0 | Number of errors: 1
| INFO    | job0 | syn                  | 0 | Number of warnings: 1
| INFO    | job0 | syn                  | 0 | Finished task in 1.36s
[... 13 lines cut ...]
  File "/home/chens/venv/lib/python3.10/site-packages/siliconcompiler/scheduler/__init__.py", line 1686, in _check_nodes_status
    raise SiliconCompilerError(
siliconcompiler._common.SiliconCompilerError: These final steps could not be reached: ['write.views', 'write.gds']
```
Here is the Python script I used for compilation. Could you please advise on how to resolve this issue?
```python
# -*- coding: utf-8 -*-
import siliconcompiler
from siliconcompiler.targets import freepdk45_demo

import os

if __name__ == '__main__':
    root = os.path.dirname(__file__)
    chip = siliconcompiler.Chip ('mkDotProduct_nt_Int32')
    chip.input(os.path.join(root, 'DotProduct.bsv'))
    chip.set('option', 'ydir', [r'./'])
    chip.set('option', 'nodisplay', True)
    chip.set('option', 'quiet', True)

    chip.clock(pin='CLK', period=1)
    chip.use(freepdk45_demo)
    chip.run()
    chip.summary()
```

chenshih1, 2025-02-13:
I tried first using the bsc compiler to translate the BSV code into Verilog, and copied the missing Verilog files from bsc's library to the same directory, then compiled the Verilog code with SiliconCompiler, and it succeeded. However, this approach did not fully leverage the convenience provided by the Bluespec API offered by SC.😂😂😂

gadfort, 2025-02-13:
@chenshih1 we don't have many examples using the bluespec compiler. Would you be able to share your test case with us and we can fix the driver appropriately?

chenshih1, 2025-02-14:
I have uploaded the BSV test cases I used to [Bluespec_BSV_TestCases](https://github.com/chenshih1/Bluespec_BSV_TestCases), which include three examples. The DotProduct example is modified based on Eg04a_MicroArchs from the [Bluespec_BSV_Tutorial](https://github.com/rsnikhil/Bluespec_BSV_Tutorial), while BubbleSort and MergeSort are derived from Eg03e_Bubblesort and Eg06a_Mergesort, respectively. Each folder contains a Python script used for compilation.

Among these test cases, BubbleSort compiles successfully because it does not reference any modules from the bsc standard library. However, DotProduct and MergeSort encounter the compilation errors mentioned earlier.

The version of the bsc compiler I used is Bluespec Compiler, version 2024.07 (build b4f31db). The binary version of this compiler can be obtained from the [Bluespec Compiler](https://github.com/B-Lang-org/bsc) on GitHub.

gadfort, 2025-02-14:
@chenshih1 thanks, do you mind if I take the dotproduct example and add it to our examples and testing?

chenshih1, 2025-02-14:
@gadfort Sure, feel free to use the dot product example. I'm glad to be able to help with this project. 🙂

gadfort, 2025-02-14:
@chenshih1 thanks. I opened a PR which should allow you to compile with imported modules

PR #3214 (2025-02-14): bluespec handle referenced modules
Closes #3209
closed 2025-02-14 by gadfort