siliconcompiler/siliconcompiler #3252: _select_inputs need to return the pruned nodes

Ciprian167, 2025-02-25:
Referring to #3235 discussion and particularly this [point](https://github.com/siliconcompiler/siliconcompiler/discussions/3235#discussioncomment-12279101). 

I managed to get that issue in a test, with a modified `builtin/nop`. 

<details>

<summary>siliconcompiler/tools/builtin/nop.py</summary>

```python
from siliconcompiler.tools.builtin import _common
from siliconcompiler.tools._common import get_tool_task
from siliconcompiler.tools.builtin.builtin import set_io_files
from siliconcompiler import flowgraph

def setup(chip):
    '''
    A no-operation that passes inputs to outputs.
    '''

    set_io_files(chip)

# Commented this out, because this was being used in scheduler._select_inputs. 
# This function also changes the inputs to prune_inputs, which was preventing the error to appear. 
# def _select_inputs(chip, step, index):
#     return _common._select_inputs(chip, step, index)

def _gather_outputs(chip, step, index):
    '''Return set of filenames that are guaranteed to be in outputs
    directory after a successful run of step/index.'''

    flow = chip.get('option', 'flow')

    in_nodes = chip.get('flowgraph', flow, step, index, 'input')
    in_task_outputs = [flowgraph._gather_outputs(chip, *node) for node in in_nodes]

    if len(in_task_outputs) > 0:
        return in_task_outputs[0].union(*in_task_outputs[1:])

    return []

def run(chip):
    return _common.run(chip)

def post_process(chip):
    _common.post_process(chip)

```

</details>

<details>

<summary>tests/flows/test_prune.py</summary>

``` python

def test_input_provides_with_prune_testcase_failing():
    chip = siliconcompiler.Chip('test')
    flow_name = 'test_flow'

    flow = siliconcompiler.Flow(flow_name)
    flow.node(flow_name, 'initial', nop)
    flow.node(flow_name, 'onestep', nop)
    flow.node(flow_name, 'twostep', nop)
    flow.node(flow_name, 'finalstep', nop)
    flow.edge(flow_name, 'initial', 'onestep')
    flow.edge(flow_name, 'initial', 'twostep')
    flow.edge(flow_name, 'onestep', 'finalstep')
    flow.edge(flow_name, 'twostep', 'finalstep')

    chip.use(flow)
    chip.set('option', 'flow', flow_name)
    chip.set('tool', 'builtin', 'task', 'nop', 'output', 'test.initial.txt', step='initial', index='0')
    chip.set('tool', 'builtin', 'task', 'nop', 'output', 'test.one.txt', step='onestep', index='0')
    chip.set('tool', 'builtin', 'task', 'nop', 'output', 'test.two.txt', step='twostep', index='0')

    chip.set('option', 'prune', ('onestep', '0'))

    assert chip.run()

```
</details>

Output:

```text

| INFO    | job0 | --        | - | Setting up node initial0 with builtin/nop
| INFO    | job0 | --        | - | Setting up node onestep0 with builtin/nop
| INFO    | job0 | --        | - | Setting up node twostep0 with builtin/nop
| INFO    | job0 | --        | - | Setting up node finalstep0 with builtin/nop
| INFO    | job0 | --        | - | Checking manifest before running.
| INFO    | job0 | initial   | 0 | Running in /tmp/pytest-of-ciprian/pytest-65/test_input_provides_with_prune0/build/test/job0/initial/0
| INFO    | job0 | initial   | 0 | Finished task in 0.94s
| INFO    | job0 | twostep   | 0 | Running in /tmp/pytest-of-ciprian/pytest-65/test_input_provides_with_prune0/build/test/job0/twostep/0
| INFO    | job0 | twostep   | 0 | Finished task in 0.98s
| ERROR   | job0 | finalstep | 0 | Unable to locate outputs directory for onestep0: /tmp/pytest-of-ciprian/pytest-65/test_input_provides_with_prune0/build/test/job0/onestep/0/outputs
| ERROR   | job0 | finalstep | 0 | Halting step 'finalstep' index '0' due to errors.
| ERROR   | job0 | --        | - | These final steps could not be reached: ['finalstep']
| ERROR   | job0 | --        | - | These final steps could not be reached: ['finalstep']

```

So the tests were not capturing this error because of the `_common._select_inputs`, which is returning the `pruned nodes`. 

```python
def _select_inputs(chip, step, index):
    _, task = get_tool_task(chip, step, index)

    chip.logger.info(f"Running builtin task '{task}'")

    flow = chip.get('option', 'flow')
    return _get_pruned_node_inputs(chip, flow, (step, index))
```

The question is, shouldn't this be the default behavior in `scheduler._select_inputs`?

[image]

gadfort, 2025-02-25:
You are correct, thanks for bringing that up, I fixed the behavior and added a test to ensure the proper behavior is honored.

Ciprian167, 2025-02-25:
You're fast! Thank you.

closed 2025-02-25 by gadfort