siliconcompiler/siliconcompiler #3256: Propagate ENV variables when version checking and more

Ciprian167, 2025-02-26:
Some tools require environment variables to be set even when doing a simple version checking, sadly. Some are looking for PATH to be set, and others for extra env vars. Can we propagate the envs to `subprocess.run` calls?

```python
def _check_tool_version(chip, step, index, run_func=None):
    '''
    Check exe version
    '''

    # ...

    envvars = {}
    path = chip.get('tool', tool, 'path', step=step, index=index)
    if path:
        envvars['PATH'] = path + os.pathsep + os.environ['PATH']
    for key in chip.getkeys('tool', tool, 'task', task, 'env'):
        val = chip.get('tool', tool, 'task', task, 'env', key, step=step, index=index)
        if val:
            envvars[key] = val

    if exe is not None:
        exe_path, exe_base = os.path.split(exe)
        if veropt:
            cmdlist = [exe]
            cmdlist.extend(veropt)
            proc = subprocess.run(cmdlist,
                                  stdin=subprocess.DEVNULL,
                                  stdout=subprocess.PIPE,
                                  stderr=subprocess.STDOUT,
                                  universal_newlines=True,
                                  env=envvars)
    # ...

```

There are probably other places that we need to propagate these envs, since I'm still getting errors when running the tool, but I couldn't find them. 

A change like this would enable me to write self-contained  workflows and allow SC to fully handle the dependency management. Rather than have quirky bash scripts to setup the env.

Ciprian167, 2025-02-26:
You probably know the best way to do it. I just copy pasted some piece of code I found in other function. But I guess that could be a simple utility function.

Ciprian167, 2025-02-26:
That did it for me. Thanks.

gadfort, 2025-02-26:
Closed with https://github.com/siliconcompiler/siliconcompiler/pull/3257

closed 2025-02-26 by gadfort