siliconcompiler/siliconcompiler #3395: Openroad silent error

aolofsson, 2025-04-02:
### Version

0.32.2

### On which OS did this happen?

Linux

### On which Python version did this happen?

Python 3.12

### Reproduction Steps

see tutorial branch

python tutorial.py

### Expected Behavior

Error out on user command line error.

Currently an incorrect command line goes through silently. I had to delet the build directory to bypass the caching mechanism.

### Actual Behavior

| INFO    | job0 | apr | 0 | openroad -c
| INFO    | job0 | apr | 0 | OpenROAD v2.0-18739-g6645a5370
| INFO    | job0 | apr | 0 | Features included (+) or not (-): +GPU +GUI +Python
| INFO    | job0 | apr | 0 | This program is licensed under the BSD-3 license. See the LICENSE file for details.
| INFO    | job0 | apr | 0 | Components of this program may be licensed under more restrictive licenses which must be honored.
| INFO    | job0 | apr | 0 | Usage: /home/aolofsson/.local/bin/openroad [-help] [-version] [-no_init] [-no_splash] [-exit] [-gui] [-threads count|max] [-log file_name] [-metrics file_name] [-no_settings] [-minimize] cmd_file
| INFO    | job0 | apr | 0 |   -help                 show help and exit
| INFO    | job0 | apr | 0 |   -version              show version and exit
| INFO    | job0 | apr | 0 |   -no_init              do not read .openroad init file
| INFO    | job0 | apr | 0 |   -threads count|max    use count threads
| INFO    | job0 | apr | 0 |   -no_splash            do not show the license splash at startup
| INFO    | job0 | apr | 0 |   -exit                 exit after reading cmd_file
| INFO    | job0 | apr | 0 |   -gui                  start in gui mode
| INFO    | job0 | apr | 0 |   -minimize             start the gui minimized
| INFO    | job0 | apr | 0 |   -no_settings          do not load the previous gui settings
| INFO    | job0 | apr | 0 |   -python               start with python interpreter [limited to db operations]
| INFO    | job0 | apr | 0 |   -log <file_name>      write a log in <file_name>
| INFO    | job0 | apr | 0 |   -metrics <file_name>  write metrics in <file_name> in JSON format
| INFO    | job0 | apr | 0 |   cmd_file              source cmd_file
| INFO    | job0 | apr | 0 | openroad>
| WARNING | job0 | apr | 0 | OpenROAD metrics file is missing
| INFO    | job0 | apr | 0 | Number of errors: 0
| INFO    | job0 | apr | 0 | Number of warnings: 0
| INFO    | job0 | apr | 0 | Finished task in 2.15s
(venv) $ sc  -versom

gadfort, 2025-04-02:
The issue here is that openroad doesn't report a non-zero exit code

gadfort, 2025-04-06:
Fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/6969

closed 2025-04-06 by gadfort