siliconcompiler/siliconcompiler #3467: CLI Dashboard: end of run information

gadfort, 2025-04-25:
From: https://github.com/siliconcompiler/siliconcompiler/issues/3463

Also, do you think it makes sense that on the last rendering to show the entire error list, outside of the live display. That would allow the user to scroll through all the errors without them being truncated by the live display.

Ciprian167, 2025-06-23:
@gadfort I've been testing the CLI dashboard on projects that have large number of tests. When there are many tests failing printing everything would get very messy. I think for the `job board` printing the nodes in order would be sufficient. If there is an error, print starting from the error, or print the errors based on the graph order should be enough. Which I think you already do. 

But for the complete report one solution would be to have a run summary in the `builddir`. I think that can be separated from the CLI and can be beneficial for flows that don't use the CLI dash. I think the information is already stored in the manifests, but it would be nice to have it presented in a human redable form. Have like a session summary, with that jobs failed, pointers to the log and so on. Then maybe display a path to this log at the end of the cli dashboard?

Would something like this make sense?

gadfort, 2025-07-31:
@Ciprian167 sorry, got distracted and missed this. I think it could be interesting to have an end of run report, it might make sense to wait a bit since we are working towards cleaning up the APIs and this would be easier to add afterwards.

gadfort, 2026-02-19:
@Ciprian167 I've been thinking about this recently. I have a basic thing on this branch if you want to take a look:
https://github.com/siliconcompiler/siliconcompiler/tree/eor-summary

I would be interested to know what you think would be helpful to add / remove.
I would love to find a solution to highlight some metrics on some nodes up top, but thats going to require a little more infrastructure  to support.

Ciprian167, 2026-02-20:
@gadfort Sorry. I missed this. I'll take a look.