siliconcompiler/siliconcompiler #3506: Better runtime performance on large flowgraphs

Ciprian167, 2025-05-05:
I'm seeing quite bad performance on larger graphs. I'm running a typical DV flow with `compile`, `elaborate` and `test` steps, with the test step having around 320 indices. In this large graph a rerun takes around `160s`. I'm not changing anything between runs, so all those `~160s` seems to be spend inside siliconcompiler to check the graph and so on. 

I enabled `cProfile` in the `scheduler/__init__.py:run`. From what I can see there are two functions that consume the most time:
- ` _reset_flow_nodes`: ~50.6s
- `_local_process`: >100s

Going further down the stack, there seems to be two obvious things that stand out. 

1. `flowgraph.py:99(_get_flowgraph_node_outputs)` - takes about `52s` and is called about `7800` times. Which raises the question, can we cache some of these get values? For a quick test, I modified the function to build a local cache, and reduced it to `~3s`. 
2. `schema_obj.py:71(__init__)` - takes about `>91.7s` and called `121` times. Would it be possible to reduce call number to this? 

Here are the cProfile reports:
[reset_flow_nodes.txt](https://github.com/user-attachments/files/20034743/reset_flow_nodes.txt)
[local_process.txt](https://github.com/user-attachments/files/20034744/local_process.txt)

Ciprian167, 2025-05-05:
The thing I tried in `_get_flowgraph_node_outputs` is:

```python
# >>>
_cached_flowgraph_node_outputs = {}
def _get_flowgraph_node_outputs(chip, flow, node):
    # >>>
    key = (id(chip), flow, node)
    if key in _cached_flowgraph_node_outputs:
        return _cached_flowgraph_node_outputs[key]

    node_outputs = []
    iter_nodes = _get_flowgraph_nodes(chip, flow)
    for iter_node in iter_nodes:
        iter_node_inputs = chip.get('flowgraph', flow, *iter_node, 'input')
        if node in iter_node_inputs:
            node_outputs.append(iter_node)
   # >>>
    _cached_flowgraph_node_outputs[key] = node_outputs

    return node_outputs

```

gadfort, 2025-05-05:
@Ciprian167 do you think it would be possible to add a version of your setup to `scripts/profile_sc.py`? That will help ensure we have a testcase for profiling this going forward.

I'm in the process of changing the way we handle the schema which `should` help the `schema_obj.py __init__` calls once complete. Along the way I expect to cleanup the flowgraph handling, since I agree that the information could be precomputed instead of called for 7800 times, but caching the value at the moment is not safe since it's technically possible to load a new graph with the same name and nodes, but different connectivity (odd example, but would cause a lot of issues.)

Maybe if you set `_cached_flowgraph_node_outputs` to `None` and at the start of running, you switch it to `{}`, then we can cache when the graph cannot be modified and restore `_cached_flowgraph_node_outputs` to None at the end to avoid caching in other parts of the code where that would be an issue?

Ciprian167, 2025-05-05:
I will add a version of the script. I need some time since I will have limited access to a computer this week. 

> Maybe if you set _cached_flowgraph_node_outputs to None and at the start of running, you switch it to {}, then we can cache when the graph cannot be modified and restore _cached_flowgraph_node_outputs to None at the end to avoid caching in other parts of the code where that would be an issue?

That wasn't a proposed solution. It's something quick and dirty I tried to see if it makes any difference in runtime. I was sure it will  have unexpected consequences. There are probable better ways the handle the "caching" and I'm sure you know best. 

BTW, is there a reason why there isn't a flowgraph class?