siliconcompiler/siliconcompiler #3535: Chip.summary output file

philiprbrenan, 2025-05-10:
### Feature Description

Please could chip.summary() write its helpful summary to files in both json and text format and print where it wrote these useful outputs?

See:: https://docs.siliconcompiler.com/en/latest/reference_manual/core_api.html#siliconcompiler.Chip.summary

Per: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/3159#issuecomment-2868120468

gadfort, 2025-05-12:
What format would the json be? Or csv?

philiprbrenan, 2025-05-12:
May I suggest that it match the summary page as closely as possible?  The summary page is very useful because it helps track the progress of a design - one can see if one's efforts over time are having a positive effect or not via previously saved summary pages which can be graphed to show trends.  It is of course possible to go and dig this information out of the run logs and json files for one self or to scrape it off the screen each time, but this creates effort that is being unnecessarily duplicated by every user that wants to track progress in this manner.  In summary: the summary page is useful and the ability to process it in a machine readable format, csv or json would make it even more useful especially if were easy to identify where the summary was located.

closed 2025-10-21 by gadfort