siliconcompiler/siliconcompiler #3598: Add tools for tweaking/patching inputs before import

infinitymdm, 2025-05-21:
## Summary

Add a method to "patch" or otherwise make changes to design sources (or chip "inputs" to use the siliconcompiler lingo) during setup, i.e. before starting a flow.

## Background

This issue is mostly motivated by my difficulties using siliconcompiler with an open-source hardware design (openhwgroup/cvw) optimized for proprietary tools. There were several small fixes or configuration changes we had to make to the source code in order to get the design remotely working, and since I'm not a code-owner on the design repository we had to use a fork.

> Ideally I'd want some way to tweak the source after cloning, perhaps by applying git patches to the repository so that upstream can still be the ultimate source of truth for the design. This fits in well with the idea of SC as a self-contained build system: it's pretty common for packagers to patch projects for their environment before passing to the compiler. (See just about any Arch PKGBUILD, [like this one for yosys](https://gitlab.archlinux.org/archlinux/packaging/packages/yosys/-/blob/main/PKGBUILD?ref_type=heads#L26), for an example)
> 
> _Originally posted by @infinitymdm in https://github.com/siliconcompiler/scgallery/issues/207#issuecomment-2895673560_

infinitymdm, 2025-05-21:
After spending a bit of time browsing the siliconcompiler source code, I'm thinking the simplest way to do this is to add a `patches` kwarg to the `chip.register_sources` method. The idea here is to pass an ordered list of path-like objects when calling `register_sources`. Each object in the list should be a git-format patchfile that can be applied to the associated source when cloned. 

It's pretty clear how to do this for git sources, and should be really straightforward to implement with `gitpython` (which I see siliconcompiler already uses in `siliconcompiler.package.git`). Just make a `git.Repo` object, iterate over the patchfiles, and call the `.apply` method on the repo with the patchfile after the repository is cloned. 

This feature should probably be limited exclusively to git sources, as it doesn't make much sense to me to patch files that the user already owns and can tweak at will.

The one thing I'm not clear on is how the schema would need to be adjusted to handle this. It seems to me that alongside the `path` (and optional `ref`) for a source, you would also need to store the `patches` that get applied to that source once it's local. There also would need to be triggers to apply the patches after cloning, as well as some way to check whether patches have already been applied to cached sources. (I suspect there are already mechanisms for this last item, I just don't know the SC source well enough yet to be sure.)

I'd like to hear thoughts from stakeholders on this; if a PR for this is not likely to be accepted, it would be good to know that my time would be better spent elsewhere.

infinitymdm, 2025-05-22:
Ok, that gives us some (very rough) requirements:

- Siliconcompiler shall accept patch sources from the user
    - Ideally, the mechanism should be similar to `chip.input`, but this requirement is not meant to specify what that interface looks like
- Tools for patching inputs shall work on any type of package
    - git repos
    - extracted tarballs
    - downloaded files
- Patch sources shall be in a standardized format
- Patched files shall be owned by the flow, not the package cache 
    - I.e. we don't want to patch cached files in case other projects make use of that cache

Any thoughts on where patching should happen in the flows? Probably within or right before the import step?

gadfort, 2025-05-23:
It makes sense to me to handle this during the initial phases of the setup, so it happens right before any nodes execute.

gadfort, 2025-06-20:
@infinitymdm because we are in the process of making some larger changes, I think this makes sense to delay until after those are in (~July+).