siliconcompiler/siliconcompiler #4041: Is the `asap7_demo` target usable as of now?

UditDey, 2025-08-27:
Hi, this is just a usage question I have, not a bug. Also I'm relatively inexperienced with silicon implementation so excuse me if this is naive.

For a university project I'm implementing some open source RISC-V CPUs on `asap7` using siliconcompiler. Before this I've only ever used `sc` for really small demo RTLs so this is the first time I'm using it for something "major"

I've implemented [CORE-V Wally](https://github.com/openhwgroup/cvw/) on `asap7_demo` and `freepdk45_demo` and the results are a bit surprising, the `fmax` is lower on `asap7` than `freepdk45_demo`, even though it should be expected that the 7nm process should give better performance (atleast enough to be tangible)

Going through the logs it seems that with `asap7_demo` the `cts.repair_timing` stage can't achieve any reduction in WNS even after hundreds of iters. From that point on WNS remains unchanged. Also the critical path itself is different between the 2, so I suppose synthesis mapping differences are playing a role too.

I read in the `sc` docs that the `asap7_demo` target is under development so perhaps tuning issues are to be expected, but is that the case still? I'm using `siliconcompiler 0.34.3` from pypi and I can see that quite a few things have changed in the project since that release

So in short, is my unexpected results because of `asap7_demo` being immature? Or is it a separate issue entirely?

UditDey, 2025-08-29:
Hi, thanks for the response

Yes I'm understand that both the PDKs are fake but this behavior still felt a little too odd. For more background, this is for a uni project where I'm demonstrating physical design concepts, and showing that an advanced node improves perf is kinda important here

Can I restrict which `asap7` corner is being used? In my previous experiments (with handrolled yosys + openroad) I just stuck to RVT_TT which seemed good enough for me

Also worth adding that I found [this](https://github.com/The-OpenROAD-Project/asap7/issues/16), so maybe its just a synthesis level issue (yosys can produce a better netlist on `freepdk45` maybe?)

As for wally, initially I saw the critical path within the divider. Reducing the `IDIV_BITSPERCYCLE` param changed it to somewhere else in the IEU datapath and it increased fmax, but was only slightly faster than `freepdk45`, so the issue remains. I'm not really sure how else I can debug it.

Also (this might be an embarrassing question), this isn't a units issue right? Because all the `asap7` SDCs I see in the `scgallery` repo you linked have clock periods that seem...pretty large, so is the units here in ns? or ps?

gadfort, 2025-08-29:
@mi-and-core thanks, yes that is correct.

@UditDey if you just want to use the TT corner you could take this file:
https://github.com/siliconcompiler/siliconcompiler/blob/v0.34.3/siliconcompiler/targets/asap7_demo.py
Create your own target with the scenarios limited to just typical:
```
    chip.set('constraint', 'timing', 'typical', 'libcorner', 'typical', clobber=False)
    chip.set('constraint', 'timing', 'typical', 'pexcorner', pex_corner, clobber=False)
    chip.set('constraint', 'timing', 'typical', 'mode', 'func', clobber=False)
    chip.set('constraint', 'timing', 'typical', 'check', ['setup', 'hold', 'power'], clobber=False)
```

gadfort, 2025-08-31:
The auto generated SDC file handles the unit conversion, but you can always verify that by looking the clock report or the saved SDC files in the output directories.