siliconcompiler/siliconcompiler #4207: asic_demo -remote fails (server error 500)

osamimi, 2025-09-30:
### Version

0.34.3

### On which OS did this happen?

Linux

### On which Python version did this happen?

Python 3.12

### Reproduction Steps

Per quick started guide, run `python3 -m siliconcompiler.demos.asic_demo -remote` on ubuntu 24 amd64

### Expected Behavior

finish the run

### Actual Behavior

```
asamimi@ubuntu:~$ python3 -m siliconcompiler.demos.asic_demo -remote

 ____  _ _ _                    ____                      _ _
/ ___|(_) (_) ___ ___  _ __    / ___|___  _ __ ___  _ __ (_) | ___ _ __
\___ \| | | |/ __/ _ \| '_ \  | |   / _ \| '_ ` _ \| '_ \| | |/ _ \ '__|
 ___) | | | | (_| (_) | | | | | |__| (_) | | | | | | |_) | | |  __/ |
|____/|_|_|_|\___\___/|_| |_|  \____\___/|_| |_| |_| .__/|_|_|\___|_|
                                                   |_|

Authors: A. Olofsson, W. Ransohoff, N. Moroze, Z. Yedidia, M. Giacometti, K. Talaei, P. Gadfort, A. Teng, P. Grossmann, G. Aguirre, M. Troiber, C. Antoci
Version: 0.34.3

--------------------------------------------------------------------------------
| INFO     | Command line argument entered: [option,remote] Value: true
Status    Node                                         Time   Warnings   Errors   Log
─────────────────────────────────────────────────────────────────────────────────────
PENDING   heartbeat/job0/elaborate/0
PENDING   heartbeat/job0/synthesis/0
PENDING   heartbeat/job0/floorplan.init/0
PENDING   heartbeat/job0/floorplan.macro_placement/0
PENDING   heartbeat/job0/floorplan.tapcell/0
PENDING   heartbeat/job0/floorplan.power_grid/0
PENDING   heartbeat/job0/floorplan.pin_placement/0
PENDING   heartbeat/job0/place.global/0
PENDING   heartbeat/job0/place.repair_design/0
PENDING   heartbeat/job0/place.detailed/0
PENDING   heartbeat/job0/cts.clock_tree_synthesis/0
PENDING   heartbeat/job0/cts.repair_timing/0
PENDING   heartbeat/job0/cts.fillcell/0
PENDING   heartbeat/job0/route.global/0
PENDING   heartbeat/job0/route.antenna_repair/0
PENDING   heartbeat/job0/route.detailed/0
PENDING   heartbeat/job0/dfm.metal_fill/0
PENDING   heartbeat/job0/write.gds/0
PENDING   heartbeat/job0/write.views/0

Progress (heartbeat/job0):  0/19 ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0%  0.0s

 | INFO     | job0 | remote               | - |   - Submitted designs must be open source. SiliconCompiler is not responsible for any
 | INFO     | job0 | remote               | - |     proprietary IP that may be uploaded.
 | INFO     | job0 | remote               | - |   - Only send one run at a time (or you may be temporarily blocked).
 | INFO     | job0 | remote               | - |   - Do not send large designs (machines have limited resources).
 | INFO     | job0 | remote               | - |   - Nontrivial designs may take some time to run. (machines have limited resources).
 | INFO     | job0 | remote               | - |   - For full TOS, see https://www.siliconcompiler.com/terms
 | INFO     | job0 | remote               | - | -------------------------------------------------------------------------------------
 | INFO     | job0 | remote               | - | Your job will be uploaded to a public server in 5 seconds.
 | ERROR    | job0 | remote               | - | Exception raised: RuntimeError / Server responded with 500:
 <!doctype html>
 <html lang="en">
 <head>
   <title>Server Error (500)</title>
 </head>
 <body>
   <h1>Server Error (500)</h1><p></p>
 </body>
 </html>
 | ERROR    | job0 | remote               | - | Traceback (most recent call last):
 | ERROR    | job0 | remote               | - |   File "<frozen runpy>", line 198, in _run_module_as_main
 | ERROR    | job0 | remote               | - |   File "<frozen runpy>", line 88, in _run_code
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/demos/asic_demo.py", line 41, in <module>
 | ERROR    | job0 | remote               | - |     proj.run()
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/project.py", line 608, in run
 | ERROR    | job0 | remote               | - |     scheduler.run()
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/scheduler/scheduler.py", line 229, in run
 | ERROR    | job0 | remote               | - |     self.run_core()
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/remote/client.py", line 35, in run_core
 | ERROR    | job0 | remote               | - |     Client(self.project).run()
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/remote/client.py", line 492, in run
 | ERROR    | job0 | remote               | - |     self.__request_run()
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/remote/client.py", line 555, in __request_run
 | ERROR    | job0 | remote               | - |     resp = self.__post('/remote_run/', post_action, success_action)
 | ERROR    | job0 | remote               | - |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 | ERROR    | job0 | remote               | - |   File "/Users/asamimi/git/siliconcompiler/siliconcompiler/remote/client.py", line 209, in __post
 | ERROR    | job0 | remote               | - |     raise RuntimeError(f'Server responded with {code}: {msg}')
```

osamimi, 2025-09-30:
Thanks! This was an editable install but it was on the git tag 0.34.3 not HOT. 
Perhaps the error message can be improved or the docs can mention that editable installs can't talk to the server?

gadfort, 2025-09-30:
normally the server would not have an issue, however, between the current version 0.34.3 and now there are a significant number of changes that, including how SC sends data to to the server. But I agree, the error message should be better (i'll leave this open, until the server had been corrected).